Kwonelee/iStoreOS-Actions #21: oect 硬件版本1.0刷完后网卡无法驱动

qtjay, 2025-09-14:
绿灯亮了，网卡灯在闪，但是没有数据包传输，也获取不到ip

qtjay, 2025-09-14:
ophub 的openwrt_rockchip_wxy-oect_k6.1.115_2025.09.10是正常的，是路由模式。刷网上7.28G容量的wxy-OEC-iStoreOS.也是正常的，是旁路由模式

Kwonelee, 2025-09-14:
> 绿灯亮了，网卡灯在闪，但是没有数据包传输，也获取不到ip 刷ophub 的openwrt_rockchip_wxy-oect_k6.1.115_2025.09.10是正常的，是路由模式。刷网上7.28G容量的wxy-OEC-iStoreOS.也是正常的，是旁路由模式

https://github.com/Kwonelee/iStoreOS-Actions/issues/5
这位同学是你这个设备wxy-oect，他刷入没有问题，你可以请教下他。另外针对wxy-oect好像O大提供了改内存还是改CPU的版本，名称为wxy-oect-replaced😄

qtjay, 2025-09-14:
感谢大佬回复，我自己判断有可能是硬件版本问题，我另外一台oect硬件2.1(电路板上印刷oec L2.1)的刷这个istoreos_24.10.2_rockchip_wxy-oect_k6.1.115_2025.09.10.img.gz就完全没问题，硬件1.0的就是上述这种网卡连不上的情况。我问一下他硬件是什么版本吧！

Kwonelee, 2025-09-16:
> 短接刷如这个[https://github.com/Kwonelee/iStoreOS-NDK/releases/download/2025.09.15-00.28-armv8/istoreos-rockchip-armv8-wxy_oec-turbo-squashfs-sysupgrade.img.gz对吗？还需不需要做其他的，比如提供日志什么的？](https://github.com/Kwonelee/iStoreOS-NDK/releases/download/2025.09.15-00.28-armv8/istoreos-rockchip-armv8-wxy_oec-turbo-squashfs-sysupgrade.img.gz%E5%AF%B9%E5%90%97%EF%BC%9F%E8%BF%98%E9%9C%80%E4%B8%8D%E9%9C%80%E8%A6%81%E5%81%9A%E5%85%B6%E4%BB%96%E7%9A%84%EF%BC%8C%E6%AF%94%E5%A6%82%E6%8F%90%E4%BE%9B%E6%97%A5%E5%BF%97%E4%BB%80%E4%B9%88%E7%9A%84%EF%BC%9F)

嗯，测试看是否能正常启动。IP默认DHCP

qtjay, 2025-09-16:
> > 短接刷如这个[https://github.com/Kwonelee/iStoreOS-NDK/releases/download/2025.09.15-00.28-armv8/istoreos-rockchip-armv8-wxy_oec-turbo-squashfs-sysupgrade.img.gz对吗？还需不需要做其他的，比如提供日志什么的？](https://github.com/Kwonelee/iStoreOS-NDK/releases/download/2025.09.15-00.28-armv8/istoreos-rockchip-armv8-wxy_oec-turbo-squashfs-sysupgrade.img.gz%E5%AF%B9%E5%90%97%EF%BC%9F%E8%BF%98%E9%9C%80%E4%B8%8D%E9%9C%80%E8%A6%81%E5%81%9A%E5%85%B6%E4%BB%96%E7%9A%84%EF%BC%8C%E6%AF%94%E5%A6%82%E6%8F%90%E4%BE%9B%E6%97%A5%E5%BF%97%E4%BB%80%E4%B9%88%E7%9A%84%EF%BC%9F)
> 
> 嗯，先测试看是否能正常启动。IP默认DHCP
试了，无法启动，刷完后直接启动就是maskroom。